mongoose-os-libs/pppos #1: PPPoS Library cannot handle a controller reset

DrBomb, 2017-11-20:
The PPPoS library cannot handle a controller reset, it connects the first time correctly.

I'm using the Adafruit Fona 3G Modem. [Link.](https://learn.adafruit.com/adafruit-fona-3g-cellular-gps-breakout/overview)

Here's a log of the connection: [Link.](https://pastebin.com/Fk2M6vyn)

The process was the following:

- Line 2 to 12: The Adafruit Modem has to be turned ON manually so the controller didn't have any answers.
- Line 12 to 36: The modem is turned on, AWS connects, SNTP is queried.
- Line 37: The ESP32 is reset.
- Line 38 onwards: mOS continues booting, it cannot continue the PPP connection even though the modem is turned on.

Surely the ESP32 will need to issue the "+++" escape command if the first AT command fails

rojer, 2017-11-20:
> Surely the ESP32 will need to issue the "+++" escape command if the first AT command fails

and that's exactly what it does, [here](https://github.com/mongoose-os-libs/pppos/blob/363167e5be92765485c6b3837f6f3e69ae46a188/src/mgos_pppos.c#L192).

Fona 3G is based on SIM5320A, i tested with SIM5320E but i don't think that is a significant difference.
i observed that sometimes it takes more than one attempt to get the modem from data and into command mode, so - what if you just wait for a while? it may eventually work.
please set `debug.level=3` and `pppos.hexdump_enable=true`, this will log all the data sent and received in the serial interface. this may help us troubleshoot.

DrBomb, 2017-11-20:
The only issue I see then is that [here](https://github.com/mongoose-os-libs/pppos/blob/363167e5be92765485c6b3837f6f3e69ae46a188/src/mgos_pppos.c#L57), the "+++" sequence is sent with a "\r\n" in the end. While the SIM5320A requires <1SecPause>+++<1SecPause> with no other characters in order to escape PPPoS.

rojer, 2017-11-20:
i had a special check at first that wouldn't send `\r\n` after `+++`, but then tested without it and it worked. but let me double-check.

rojer, 2017-11-20:
ok, a sent a change that reliably takes the device out of existing PPP session. it will be published soon and attached to this issue.

closed 2017-11-20 by rojer

rojer, 2017-11-20:
there, it's been published. @DrBomb , please try and let me know if it works for you.

DrBomb, 2017-11-20:
It works! Thank you

alvarowolfx, 2017-11-24:
Worked for me too, I'm using the SIM800L modem and now when I reset the device it also successfully reconnect with the modem. :+1: