mongoose-os-libs/pppos #2: Doesn't work with A7 module

i5okie, 2017-11-30:
Hi,
This doesn't seem to work with A7 module.

rojer, 2017-11-30:
link to module docs please.
and log with `debug.level=3`.

i5okie, 2017-11-30:
Hi,
I'm using A7 module that is designed into maduino board:  http://www.robotshop.com/en/maduino-a7-gprs-gsm-module.html

I've tried using arduino sketches off the arduino chip on the board and didn't really get anywhere.

*What i've done: *
I've let the arduino chip to turn the modem on via the pins, but uart from arduino is disconnected.
I've bypassed the arduino chip, and connected straight to the uart of the A7 chip itself.
I've confirmed that the A7 is operational with my sim card by using AT Command Tester (http://m2msupport.net/m2msupport/download-at-command-tester/) through FTDI usb to serial adapter. I can confirm that i'm on the network, good enough signal, registered, can send sms, can send http post request just via at commands.

I've sniffed the uart communication between ESP32 board (using feather Huzzah32) and the A7 chip.
Mongoose OS is sending ATZ command, A7 replies with CME Error.
mos sends +++ command and A7 replies with OK.

From what I've gathered A7 does not support the ATZ command, but it does support the other commands in the pppos library (ath, ata1, etc..).
A7 for some reason adds an extra new line character after each response.
example:

AT

OK
AT

OK

manual:
https://www.smart-prototyping.com/image/data/9_Modules/101756%20Wireless%20A6C/A6_A7_A6C_datasheet-EN.pdf

I'll do the debug.level=3 later when home.

rojer, 2017-11-30:
@i5okie hm. we use ATZ to remove common settings to some baseline, but it's by no means essential. if you remove ATZ, does it work?

i5okie, 2017-12-01:
@rojer  I'm not actually sure how to to test that.
Do I have to build locally with docker to do that?

i5okie, 2017-12-01:
I've forked this lib and change the file, but I have no clue where to put it. Or where these files are before it gets flashed.

i5okie, 2017-12-01:
okay so changed the ATZ to AT command.
now its timing out on ATH command..

[Dec  1 11:49:57.598] mgos_net_on_change_c PPP: connecting
[Dec  1 11:49:57.647] mgos_pppos_uart_disp << 5
[Dec  1 11:49:57.652] mgos_pppos_uart_disp << ATH
[Dec  1 11:49:57.661] mgos_pppos_uart_disp << 17
[Dec  1 11:49:57.661] mgos_pppos_uart_disp << +CME ERROR:58
[Dec  1 11:50:00.657] mgos_pppos_uart_disp Command timed out: ATH
[Dec  1 11:50:00.662] mgos_pppos_set_state 3 -> 0
[Dec  1 11:50:00.673] mgos_pppos_set_net_s 1 -> 0
[Dec  1 11:50:00.673] mgos_pppos_uart_disp Connecting (UART2, APN 'internet.com')...
[Dec  1 11:50:00.677] mgos_pppos_set_state 0 -> 1
[Dec  1 11:50:00.682] mgos_net_on_change_c PPP: disconnected
[Dec  1 11:50:01.686] mgos_pppos_set_state 1 -> 2
[Dec  1 11:50:01.690] mgos_pppos_at_cmd    >> +++
[Dec  1 11:50:01.745] mgos_pppos_uart_disp << 3
[Dec  1 11:50:02.948] mgos_pppos_set_net_s 0 -> 1
[Dec  1 11:50:02.948] mgos_pppos_set_state 2 -> 3
[Dec  1 11:50:02.953] mgos_pppos_at_cmd    >> ATH

i5okie, 2017-12-01:
of course it'd help to put the sim card in.
```
[Dec  1 11:57:24.254] mgos_net_on_change_c PPP: connecting
[Dec  1 11:57:24.303] mgos_pppos_uart_disp << 5
[Dec  1 11:57:24.308] mgos_pppos_uart_disp << ATH
[Dec  1 11:57:24.316] mgos_pppos_uart_disp << 17
[Dec  1 11:57:24.316] mgos_pppos_uart_disp << +CME ERROR:58
[Dec  1 11:57:24.939] mgos_pppos_uart_disp << 31
[Dec  1 11:57:24.946] mgos_pppos_uart_disp << +CTZV:17/12/01,19:57:24,+04
[Dec  1 11:57:25.779] mgos_pppos_uart_disp << 41
[Dec  1 11:57:25.788] mgos_pppos_uart_disp << +CIEV: service,  1
[Dec  1 11:57:25.788] mgos_pppos_uart_disp << +CIEV: roam, 0
[Dec  1 11:57:25.796] mgos_pppos_uart_disp << 9
[Dec  1 11:57:25.796] mgos_pppos_uart_disp << +CREG: 1
[Dec  1 11:57:27.292] mgos_pppos_uart_disp Command timed out: ATH
[Dec  1 11:57:27.297] mgos_pppos_set_state 3 -> 0
[Dec  1 11:57:27.307] mgos_pppos_set_net_s 1 -> 0
[Dec  1 11:57:27.307] mgos_pppos_uart_disp Connecting (UART2, APN 'internet.com')...
[Dec  1 11:57:27.312] mgos_pppos_set_state 0 -> 1
[Dec  1 11:57:27.317] mgos_net_on_change_c PPP: disconnected
[Dec  1 11:57:28.321] mgos_pppos_set_state 1 -> 2
[Dec  1 11:57:28.326] mgos_pppos_at_cmd    >> +++
[Dec  1 11:57:28.429] mgos_pppos_uart_disp << 3
[Dec  1 11:57:29.533] mgos_pppos_set_net_s 0 -> 1
[Dec  1 11:57:29.534] mgos_pppos_set_state 2 -> 3
[Dec  1 11:57:29.538] mgos_pppos_at_cmd    >> ATH
[Dec  1 11:57:29.544] mgos_net_on_change_c PPP: connecting
[Dec  1 11:57:29.593] mgos_pppos_uart_disp << 5
[Dec  1 11:57:29.597] mgos_pppos_uart_disp << ATH
[Dec  1 11:57:29.605] mgos_pppos_uart_disp << 17
[Dec  1 11:57:29.605] mgos_pppos_uart_disp << +CME ERROR:58
[Dec  1 11:57:31.565] mgos_pppos_uart_disp << 8
[Dec  1 11:57:31.570] mgos_pppos_uart_disp << RING
[Dec  1 11:57:32.566] mgos_pppos_uart_disp << 8
[Dec  1 11:57:32.572] mgos_pppos_uart_disp Command timed out: ATH
[Dec  1 11:57:32.577] mgos_pppos_set_state 3 -> 0
[Dec  1 11:57:32.587] mgos_pppos_set_net_s 1 -> 0
[Dec  1 11:57:32.587] mgos_pppos_uart_disp Connecting (UART2, APN 'internet.com')...
[Dec  1 11:57:32.592] mgos_pppos_set_state 0 -> 1
[Dec  1 11:57:32.597] mgos_net_on_change_c PPP: disconnected
[Dec  1 11:57:33.567] mgos_pppos_uart_disp << 8
[Dec  1 11:57:33.661] mgos_pppos_set_state 1 -> 2
[Dec  1 11:57:33.666] mgos_pppos_at_cmd    >> +++
[Dec  1 11:57:33.721] mgos_pppos_uart_disp << 3
[Dec  1 11:57:34.568] mgos_pppos_uart_disp << 8
[Dec  1 11:57:34.873] mgos_pppos_set_net_s 0 -> 1
[Dec  1 11:57:34.874] mgos_pppos_set_state 2 -> 3
[Dec  1 11:57:34.878] mgos_pppos_at_cmd    >> ATH
[Dec  1 11:57:34.883] mgos_net_on_change_c PPP: connecting
[Dec  1 11:57:34.935] mgos_pppos_uart_disp << 5
[Dec  1 11:57:34.940] mgos_pppos_uart_disp << ATH
[Dec  1 11:57:34.948] mgos_pppos_uart_disp << 17
[Dec  1 11:57:34.948] mgos_pppos_uart_disp << +CME ERROR:58
[Dec  1 11:57:35.570] mgos_pppos_uart_disp << 8
[Dec  1 11:57:35.574] mgos_pppos_uart_disp << RING
[Dec  1 11:57:36.571] mgos_pppos_uart_disp << 8
[Dec  1 11:57:36.576] mgos_pppos_uart_disp << RING
[Dec  1 11:57:37.571] mgos_pppos_uart_disp << 7
[Dec  1 11:57:37.576] mgos_pppos_uart_disp << RING
[Dec  1 11:57:37.580] mgos_pppos_uart_disp << 1
[Dec  1 11:57:37.882] mgos_pppos_uart_disp Command timed out: ATH
[Dec  1 11:57:37.887] mgos_pppos_set_state 3 -> 0
[Dec  1 11:57:37.892] mgos_pppos_set_net_s 1 -> 0
[Dec  1 11:57:37.899] mgos_pppos_uart_disp Connecting (UART2, APN 'internet.com')...
[Dec  1 11:57:37.904] mgos_pppos_set_state 0 -> 1
[Dec  1 11:57:37.909] mgos_net_on_change_c PPP: disconnected
[Dec  1 11:57:38.572] mgos_pppos_uart_disp << 8
[Dec  1 11:57:38.971] mgos_pppos_set_state 1 -> 2
[Dec  1 11:57:38.976] mgos_pppos_at_cmd    >> +++
[Dec  1 11:57:39.030] mgos_pppos_uart_disp << 3
[Dec  1 11:57:39.575] mgos_pppos_uart_disp << 8
[Dec  1 11:57:40.274] mgos_pppos_set_net_s 0 -> 1
[Dec  1 11:57:40.274] mgos_pppos_set_state 2 -> 3
[Dec  1 11:57:40.278] mgos_pppos_at_cmd    >> ATH
[Dec  1 11:57:40.283] mgos_net_on_change_c PPP: connecting
[Dec  1 11:57:40.333] mgos_pppos_uart_disp << 5
[Dec  1 11:57:40.338] mgos_pppos_uart_disp << ATH
[Dec  1 11:57:40.345] mgos_pppos_uart_disp << 17
[Dec  1 11:57:40.345] mgos_pppos_uart_disp << +CME ERROR:58
[Dec  1 11:57:40.576] mgos_pppos_uart_disp << 8
[Dec  1 11:57:40.581] mgos_pppos_uart_disp << RING
[Dec  1 11:57:41.578] mgos_pppos_uart_disp << 8
[Dec  1 11:57:41.582] mgos_pppos_uart_disp << RING
[Dec  1 11:57:42.579] mgos_pppos_uart_disp << 8
[Dec  1 11:57:42.584] mgos_pppos_uart_disp << RING
[Dec  1 11:57:43.282] mgos_pppos_uart_disp Command timed out: ATH
```

rojer, 2017-12-02:
well, this looks like complete mayhem to me :)
can you come up with a valid sequnce of commands to initialize this module? we'll see what we can do then.

as to how to use a modified lib. yes, you need to be building locally (deps are not sent to remote builder). then, in your app, create a `deps` folder, clone the lib's repo inside it so that you have `deps/pppos`, and ``mos will pick up the modified lib.

i5okie, 2017-12-03:
Alright so I'm not sure what PPP commands are like but here's how to get it to go into TCP connection:

```
AT+CGDCONT=1, "IP", "internet.com"
AT+CGATT=1
AT+CGACT=1,1
AT+CIPSTART="TCP", "myserver.ofl.onl", 80
AT+CIPSEND=80
>POST http://myserver.ofl.onl/itworks=yes HTTP1.0\r\n HOST:two.ofl.onl\r\n\r\n
HOST:myserver.ofl.onl
```
on myserver was listening for http requests:
"XX.XXX.XX.XXX - - [03/Dec/2017 00:49:29] code 400, message Bad request syntax ('POST http://myserver.ofl.onl/itworks=yes HTTP1.0\\r\\n HOST:myserver.ofl.onl\\r\\n\\r\\n')`

and then similar things to get it into PPP mode..
```
AT+CGDCONT=1,"PPP"
ATDT*99***1#
```
and then I see
CONNECT

rojer, 2017-12-03:
ok, i think we might be better off just making entire setup sequence configurable

rojer, 2017-12-03:
wait. you need to specify APN though. what if you do this:
```
AT+CGDCONT=1,"IP","internet.com"
ATDT*99***1#
```
do you get a `CONNECT`?

i5okie, 2017-12-04:
sure do!

```
at

OK
at+creg?

+CREG: 1,1

OK

+CTZV:17/12/04,02:37:58,+04

+CIEV: service,  1
+CIEV: roam, 0

+CREG: 1
AT+CGDCONT=1,"IP","internet.com"

OK
ATDT*99***1#

CONNECT
```

i5okie, 2017-12-07:
Would you be able to implement that?
I'm like a n00b at C/CPP

i5okie, 2017-12-07:
Last update to the library actually breaks the +++ ATH commands because it concatenates into "+++ATH" So it needs a new line character after +++ command otherwise it fails.

I'll try what I can.

rojer, 2017-12-07:
i understand the problem, need to think how best to fix it.

Albertogon, 2018-02-26:
I think, same problem with SIM800L.

`{"fd":2,"data":"mgos_pppos_uart_disp Command timed out: ATH\n","t":1519672195.581,"seq":34}
{"fd":2,"data":"mgos_pppos_uart_disp Connecting (UART1, APN 'Internetmas')...\n","t":1519672195.703,"seq":35}
{"fd":2,"data":"mgos_net_on_change_c PPP: disconnected\n","t":1519672197.415,"seq":36}
{"fd":2,"data":"mgos_net_on_change_c PPP: connecting\n","t":1519672199.554,"seq":37}
{"fd":2,"data":"mgos_pppos_uart_disp Command timed out: ATH\n","t":1519672201.482,"seq":38}
{"fd":2,"data":"mgos_pppos_uart_disp Connecting (UART1, APN 'Internetmas')...\n","t":1519672201.604,"seq":39}
{"fd":2,"data":"mgos_net_on_change_c PPP: disconnected\n","t":1519672202.691,"seq":40}
{"fd":2,"data":"mgos_net_on_change_c PPP: connecting\n","t":1519672205.893,"seq":41}
{"fd":2,"data":"mgos_pppos_uart_disp Command timed out: ATH\n","t":1519672207.396,"seq":42}
{"fd":2,"data":"mgos_pppos_uart_disp Connecting (UART1, APN 'Internetmas')...\n","t":1519672207.518,"seq":43}
{"fd":2,"data":"mgos_net_on_change_c PPP: disconnected\n","t":1519672209.127,"seq":44}
{"fd":2,"data":"mgos_net_on_change_c PPP: connecting\n","t":1519672211.265,"seq":45}
{"fd":2,"data":"mgos_pppos_uart_disp Command timed out: ATH\n","t":1519672213.309,"seq":46}
{"fd":2,"data":"mgos_pppos_uart_disp Connecting (UART1, APN 'Internetmas')...\n","t":1519672213.431,"seq":47}
{"fd":2,"data":"mgos_net_on_change_c PPP: disconnected\n","t":1519672215.467,"seq":48}
{"fd":2,"data":"mgos_net_on_change_c PPP: connecting\n","t":1519672217.609,"seq":49}
{"fd":2,"data":"mgos_pppos_uart_disp Command timed out: ATH\n","t":1519672219.218,"seq":50}
{"fd":2,"data":"mgos_pppos_uart_disp Connecting (UART1, APN 'Internetmas')...\n","t":1519672219.343,"seq":51}
{"fd":2,"data":"mgos_net_on_change_c PPP: disconnected\n","t":1519672220.843,"seq":52}
{"fd":2,"data":"mgos_net_on_change_c PPP: connecting\n","t":1519672223.949,"seq":53}
{"fd":2,"data":"mgos_pppos_uart_disp Command timed out: ATH\n","t":1519672225.131,"seq":54}
{"fd":2,"data":"mgos_pppos_uart_disp Connecting (UART1, APN 'Internetmas')...\n","t":1519672225.253,"seq":55}
{"fd":2,"data":"mgos_net_on_change_c PPP: disconnected\n","t":1519672227.180,"seq":56}
{"fd":2,"data":"mgos_net_on_change_c PPP: connecting\n","t":1519672229.322,"seq":57}
{"fd":2,"data":"mgos_pppos_uart_disp Command timed out: ATH\n","t":1519672231.043,"seq":58}
{"fd":2,"data":"mgos_pppos_uart_disp Connecting (UART1, APN 'Internetmas')...\n","t":1519672231.165,"seq":59}
{"fd":2,"data":"mgos_net_on_change_c PPP: disconnected\n","t":1519672232.465,"seq":60}
{"fd":2,"data":"mgos_net_on_change_c PPP: connecting\n","t":1519672235.666,"seq":61}
{"fd":2,"data":"mgos_pppos_uart_disp Command timed out: ATH\n","t":1519672236.958,"seq":62}`

This is my config:
"pppos": {
    "enable": true,
    "uart_no": 1,
    "baud_rate": 9600,
    "fc_enable": false,
    "apn": "Internetmas",
    "user": "",
    "pass": "",
    "connect_cmd": "ATD*99***1#",
    "echo_interval": 10,
    "echo_fails": 3,
    "hexdump_enable": false
  },

i5okie, 2018-02-26:
I got a different sim card. I've tested using an Arduino library which
worked fine. I'll test the this with mongoose and the new sim card. just
for sanity check. Because I couldn't properly connect using an Arduino with
A7 unless I was doing it manually.

rojer, 2018-03-03:
@Albertogon `ATH` command times out. this is the very first command, so it likely just means the device cannot communicate with the module.

i5okie, 2018-03-03:
Or that A7 doesn't accept ATH command

ndnparis, 2018-03-20:
Hi all,
I am using a SIM800L connected to a nodeMCU ESP32 and I have the same issue.

Mar 20 18:47:35.588] mgos_net_on_change_c PPP: connecting
[Mar 20 18:47:35.598] == Net event: 1 CONNECTING 
[Mar 20 18:47:38.597] mgos_pppos_uart_disp Command timed out: ATH
[Mar 20 18:47:38.603] mgos_pppos_set_state 3 -> 0
[Mar 20 18:47:38.608] mgos_pppos_set_net_s 1 -> 0
[Mar 20 18:47:38.610] mgos_pppos_uart_disp Connecting (UART1, APN 'free')...
[Mar 20 18:47:38.617] mgos_pppos_set_state 0 -> 1
[Mar 20 18:47:38.622] mgos_net_on_change_c PPP: disconnected
[Mar 20 18:47:38.630] Disconnected -------------->>> 
[Mar 20 18:47:38.635] == Net event: 0 DISCONNECTED 
[Mar 20 18:47:39.637] mgos_pppos_set_state 1 -> 2
[Mar 20 18:47:39.642] mgos_pppos_at_cmd    >> +++
[Mar 20 18:47:40.937] mgos_pppos_set_net_s 0 -> 1
[Mar 20 18:47:40.940] mgos_pppos_set_state 2 -> 3
[Mar 20 18:47:40.944] mgos_pppos_at_cmd    >> ATH
[Mar 20 18:47:40.949] mgos_net_on_change_c PPP: connecting
[Mar 20 18:47:40.959] == Net event: 1 CONNECTING 
c[Mar 20 18:47:43.957] mgos_pppos_uart_disp Command timed out: ATH
[Mar 20 18:47:43.963] mgos_pppos_set_state 3 -> 0
[Mar 20 18:47:43.968] mgos_pppos_set_net_s 1 -> 0
[Mar 20 18:47:43.972] mgos_pppos_uart_disp Connecting (UART1, APN 'free')...
[Mar 20 18:47:43.979] mgos_pppos_set_state 0 -> 1
[Mar 20 18:47:43.984] mgos_net_on_change_c PPP: disconnected
[Mar 20 18:47:43.993] Disconnected -------------->>> 
[Mar 20 18:47:43.998] == Net event: 0 DISCONNECTED 

Have you got solutions?
Thanks

sobiso, 2018-07-12:
`[Jul 12 08:10:17.866] mgos_net_on_change_c PPP: ready, IP 10.161.111.224, GW 192.168.254.254, DNS 89.108.202.21
[Jul 12 08:10:17.870] net_cb               Net got IP address
[Jul 12 08:10:17.876] mgos_bt_net_ev       Network is up, disabling Bluetooth
[Jul 12 08:10:17.969] save_cfg             Saved to conf9.json
[Jul 12 08:10:17.976] mgos_mqtt_global_con MQTT connecting to iot.eclipse.org:1883
[Jul 12 08:10:18.899] mgos_sntp_query      SNTP query to time.google.com
[Jul 12 08:10:25.667] mgos_mqtt_ev         MQTT TCP connect ok (0)
[Jul 12 08:10:28.240] mg_resolve_async_eh  Failed to resolve 'time.google.com', server 89.108.202.21
[Jul 12 08:10:32.710] mgos_sntp_query      SNTP query to time.google.com
[Jul 12 08:10:35.266] mgos_mqtt_ev         MQTT CONNACK 0
[Jul 12 08:10:35.272] do_subscribe         Subscribing to 'esp32_45834C/rpc/#'
[Jul 12 08:10:35.278] do_subscribe         Subscribing to 'esp32_45834C/rpc'
[Jul 12 08:10:40.838] led_timer_cb         Tick uptimee: 61.46, RAM: 228196, 121276 free
[Jul 12 08:10:42.240] mg_resolve_async_eh  Failed to resolve 'time.google.com', server 89.108.202.21
[Jul 12 08:10:47.831] mgos_sntp_query      SNTP query to time.google.com
[Jul 12 08:10:57.280] mg_resolve_async_eh  Failed to resolve 'time.google.com', server 89.108.202.21`

First line PPP:ready, this GW is good? I tested it on 2 simcards and the same result. Around 1/30 attempts i get connection with SNTP and MQTT

rfselim, 2018-07-14:
Same problem with SIM800L module. Did you find a solution.

[Jul 14 09:38:59.602] mgos_net_on_change_c PPP: connecting
[Jul 14 09:39:02.702] mgos_pppos_uart_disp Command timed out: ATH
[Jul 14 09:39:02.709] mgos_pppos_uart_disp Connecting (UART1, APN 'internet')...
[Jul 14 09:39:02.714] mgos_net_on_change_c PPP: disconnected
[Jul 14 09:39:05.112] mgos_net_on_change_c PPP: connecting
[Jul 14 09:39:08.212] mgos_pppos_uart_disp Command timed out: ATH
[Jul 14 09:39:08.219] mgos_pppos_uart_disp Connecting (UART1, APN 'internet')...
[Jul 14 09:39:08.225] mgos_net_on_change_c PPP: disconnected

sobiso, 2018-07-14:
Connect UART-USB and check that SIM800 response to AT command.

rfselim, 2018-07-14:
AT commands are working.

sobiso, 2018-07-14:
Check signal, AT+CSQ

rfselim, 2018-07-14:
I got this

AT+CSQ
+CSQ: 18,0

OK

sobiso, 2018-07-14:
I think we should wait for author of this lib

rojer, 2019-02-09:
please re-test with latest master, there have been major improvements.

closed 2019-02-09 by rojer