mongoose-os-libs/pppos #4: PPPoS works with open broker, but not with AWS IOT (SIM800L)

ashwinpromethean, 2018-08-18:
Hi,

I am trying to connect ESP32 to the internet using a SIM800L. I have tried connecting to a few open source brokers on port 1883 (like iot.eclipse.org), and it works well. But the moment I try to connect to AWS-IOT, it stops connecting and PPP starts disconnecting. Note that I am able to connect to AWS IOT over Wifi easily.

**The console output i get are as follows after restart :** 
[Aug 18 15:31:04.919] mgos_ota_http_client Init done, ota_lib=(null), ota handler 1
[Aug 18 15:31:04.926] mgos_ota_shadow_init OTA over shadow initialised
[Aug 18 15:31:04.932] mgos_wifi_setup      WiFi mode: off
[Aug 18 15:31:04.942] mgos_http_server_ini HTTP server started on [80]
[Aug 18 15:31:04.942] mgos_pppos_create    PPPoS UART2, 9600, fc off, APN 'internet'
[Aug 18 15:31:04.953] mgos_provision_init  Factory reset via button: pin 0 (enabled), hold_ms 0 (reset on reboot)
[Aug 18 15:31:04.973] button_timer_cb      Button released, NOT resetting
[Aug 18 15:31:04.974] mg_rpc_channel_mqtt  0x3ffb0b00 esp32_1D1CD0/rpc
[Aug 18 15:31:04.981] mg_rpc_channel_uart  0x3ffb0f74 UART0
[Aug 18 15:31:04.990] mgos_init            Init done, RAM: 226188 total, 177848 free, 175572 min free
[Aug 18 15:31:05.298] LED GPIO: 21 button GPIO: 0 
[Aug 18 15:31:05.306] mgos_aws_shadow_init Device shadow name: esp32_1D1CD0 (token b5b66125)
[Aug 18 15:31:05.313] mgos_pppos_uart_disp Connecting (UART2, APN 'internet')...
[Aug 18 15:31:05.322] mongoose_poll        New heap free LWM: 155800
[Aug 18 15:31:05.322] mgos_net_on_change_c PPP: disconnected
[Aug 18 15:31:07.722] mgos_net_on_change_c PPP: connecting
[Aug 18 15:31:10.754] mgos_pppos_uart_disp Command timed out: ATH
[Aug 18 15:31:10.754] mgos_pppos_uart_disp Connecting (UART2, APN 'internet')...
[Aug 18 15:31:10.764] mgos_net_on_change_c PPP: disconnected
[Aug 18 15:31:13.263] mgos_net_on_change_c PPP: connecting
[Aug 18 15:31:13.835] mgos_net_on_change_c PPP: connected
[Aug 18 15:31:13.835] mgos_pppos_uart_disp Starting PPP, user ''
[Aug 18 15:31:13.844] mongoose_poll        New heap free LWM: 151592
[Aug 18 15:31:16.925] mongoose_poll        New heap free LWM: 151372
[Aug 18 15:31:16.931] mongoose_poll        New heap free LWM: 151296 
[Aug 18 15:32:26.195] mgos_pppos_status_cb Error 6 (phase 0), reconnect
[Aug 18 15:32:26.195] mgos_pppos_uart_disp Connecting (UART2, APN 'internet')...
[Aug 18 15:32:26.206] mgos_net_on_change_c PPP: disconnected
[Aug 18 15:32:28.613] mgos_net_on_change_c PPP: connecting
[Aug 18 15:32:31.645] mgos_pppos_uart_disp Command timed out: ATH
[Aug 18 15:32:31.645] mgos_pppos_uart_disp Connecting (UART2, APN 'internet')...
[Aug 18 15:32:31.655] mgos_net_on_change_c PPP: disconnected
[Aug 18 15:32:34.154] mgos_net_on_change_c PPP: connecting
[Aug 18 15:32:37.226] mgos_pppos_uart_disp Command timed out: ATH
[Aug 18 15:32:37.226] mgos_pppos_uart_disp Connecting (UART2, APN 'internet')...

**The hexdump looks like this :** 
[Aug 18 15:37:06.449] mgos_pppos_uart_disp Connecting (UART2, APN 'internet')...
[Aug 18 15:37:06.459] mgos_net_on_change_c PPP: disconnected
[Aug 18 15:37:08.957] mgos_net_on_change_c PPP: connecting
[Aug 18 15:37:08.989] 0000  5d d6 7e 0d 0a 4e                                ].~..N
[Aug 18 15:37:11.990] mgos_pppos_uart_disp Command timed out: ATH
[Aug 18 15:37:11.990] mgos_pppos_uart_disp Connecting (UART2, APN 'internet')...
[Aug 18 15:37:12.000] mgos_net_on_change_c PPP: disconnected
[Aug 18 15:37:14.418] mgos_net_on_change_c PPP: connecting
[Aug 18 15:37:14.450] 0000  4f 20 43 41 52 52                                O CARR
[Aug 18 15:37:17.450] mgos_pppos_uart_disp Command timed out: ATH
[Aug 18 15:37:17.450] mgos_pppos_uart_disp Connecting (UART2, APN 'internet')...
[Aug 18 15:37:17.460] mgos_net_on_change_c PPP: disconnected
[Aug 18 15:37:19.959] mgos_net_on_change_c PPP: connecting
[Aug 18 15:37:19.990] 0000  49 45 52 0d 0a 0d                                IER...
[Aug 18 15:37:23.021] mgos_pppos_uart_disp Command timed out: ATH
[Aug 18 15:37:23.021] mgos_pppos_uart_disp Connecting (UART2, APN 'internet')...
[Aug 18 15:37:23.031] mgos_net_on_change_c PPP: disconnected
[Aug 18 15:37:25.529] mgos_net_on_change_c PPP: connecting
[Aug 18 15:37:25.561] 0000  0a 4f 4b 0d 0a 0d                                .OK...
[Aug 18 15:37:25.593] 0000  0a 4f 4b 0d 0a 0d                                .OK...
[Aug 18 15:37:25.628] 0000  0a 4f 4b 0d 0a 0d 0a 4f 4b 0d 0a                 .OK....OK..
[Aug 18 15:37:25.691] 0000  0a 0d 0a 4f 4b 0d 0a                             ...OK..
[Aug 18 15:37:25.737] 0000  0a 0d 0a 4f 4b 0d 0a 0d 0a 4f 4b 0d              ...OK....OK.
[Aug 18 15:37:25.745] mgos_pppos_uart_disp Starting PPP, user ''
[Aug 18 15:37:25.750] 0000  7e ff 7d 23 c0 21 7d 21 7d 21 7d 20 7d 34 7d 22  ~.}#.!}!}!} }4}"
[Aug 18 15:37:25.756] 0010  7d 26 7d 20 7d 20 7d 20 7d 20 7d 25 7d 26 7d 27  }&} } } } }%}&}'
[Aug 18 15:37:25.763] 0020  af 34 4b 7d 27 7d 22 7d 28 7d 22 f9 20 7e        .4K}'}"}(}". ~
[Aug 18 15:37:25.769] mongoose_poll        New heap free LWM: 150796
[... 105 lines cut ...]
[Aug 18 15:38:40.801] 0000  7d 26 7d 20 7d 2a                                }&} }*
[Aug 18 15:38:43.832] mgos_pppos_uart_disp Command timed out: ATH
[Aug 18 15:38:43.832] mgos_pppos_uart_disp Connecting (UART2, APN 'internet')...
[Aug 18 15:38:43.842] mgos_net_on_change_c PPP: disconnected
[Aug 18 15:38:46.340] mgos_net_on_change_c PPP: connecting
[Aug 18 15:38:46.371] 0000  7d 20 7d 20 96 cb                                } } ..

I have to reiterate that this works over wifi, but the moment i change to PPP, it stops connecting.

The PPPoS config looks like this
  "pppos": {
    "enable": true,
    "uart_no": 2,
    "baud_rate": 9600,
    "fc_enable": false,
    "apn": "internet",
    "user": "",
    "pass": "",
    "connect_cmd": "ATDT*99***1#",
    "echo_interval": 10,
    "echo_fails": 3,
    "hexdump_enable": true
  }

What would be the issue here?

nicoloboatto, 2019-02-09:
Did you manage to get this working, by any chance?

rojer, 2019-02-09:
a lot of work happened lately to get connection stable, closing.

closed 2019-02-09 by rojer