mongoose-os-libs/pppos #5: Connection drops when transferring larger files

nicoloboatto, 2019-02-09:
The device (an M5Stack with SIM800L module) manages to connect to the mDash and it successfully sends smaller files over such as conf9.json which is just a few lines.

Whenever I request a larger file I get these errors and from there on the connection stops working altogether. 

[Feb  9 04:54:59.525] rpc_fs_get_handler   Sending conf0.json
[Feb  9 04:55:32.785] mg_ssl_mbed_log      0x3ffd10a8 mbedtls_ssl_flush_output() returned -78 (-0x004e)
[Feb  9 04:55:32.793] mg_ssl_mbed_log      0x3ffd10a8 mbedtls_ssl_write_record() returned -78 (-0x004e)
[Feb  9 04:55:32.801] mg_ssl_mbed_log      0x3ffd10a8 mbedtls_ssl_send_alert_message() returned -78 (-0x004e)
[Feb  9 04:55:37.811] mg_rpc_channel_ws_ou 0x3ffc3954 Connecting to wss://dash.mongoose-os.com/api/v2/rpc, SSL? 1
[Feb  9 04:55:39.257] mgos_pppos_status_cb Error 9 (phase 0), reconnect
[Feb  9 04:55:39.263] mgos_pppos_dispatch_ Connecting (UART1, APN 'everywhere')...
[Feb  9 04:55:39.269] mgos_net_on_change_c PPP: disconnected
[Feb  9 04:55:41.978] mgos_net_on_change_c PPP: connecting
[Feb  9 04:55:44.989] mgos_pppos_dispatch_ Command timed out: ATH
[Feb  9 04:55:44.995] mgos_pppos_dispatch_ Connecting (UART1, APN 'everywhere')...
[Feb  9 04:55:45.001] mgos_net_on_change_c PPP: disconnected
[Feb  9 04:55:47.643] mg_resolve_async_eh  Failed to resolve 'dash.mongoose-os.com', server 8.8.8.8
[Feb  9 04:55:47.649] mg_rpc_ws_out_handle 0x3ffc3954 TCP connect failed: -1
[Feb  9 04:55:47.978] mgos_net_on_change_c PPP: connecting
[Feb  9 04:55:50.989] mgos_pppos_dispatch_ Command timed out: ATH
[Feb  9 04:55:50.995] mgos_pppos_dispatch_ Connecting (UART1, APN 'everywhere')...
[Feb  9 04:55:51.002] mgos_net_on_change_c PPP: disconnected
[Feb  9 04:55:53.978] mgos_net_on_change_c PPP: connecting
[Feb  9 04:55:53.994] mgos_pppos_gsn_cb    , IMEI: TE0

As you can see the IMEI that gets printed seems to be the response to another command, it's as if the serial connection got out of sync.

rojer, 2019-02-09:
there has been [a lot](https://github.com/cesanta/mongoose-os/compare/2.11.0...master) of [work](https://github.com/mongoose-os-libs/pppos/compare/2.11.0...master) lately specifically on connection stability. i can tell that you are using release version, please try latest master of both mongoose os and the driver - `mos update latest`. also, please use UART2 (because of [this](https://github.com/cesanta/mongoose-os/commit/bdff712ae5a86bb064b0792335e51e2f469f8f9a)).
i have tested specifically with SIM800L and the connection works fine. if it doesn't work well even on master, please reopen.

closed 2019-02-09 by rojer

nicoloboatto, 2019-02-10:
After updating both mongoose os and the driver I run into a new issue.
The serial connection to my computer completely stops working right after PPPoS gets initialised.

[Feb 10 19:50:07.711] mgos_deps_init.c:118    init pppos...
[Feb 10 19:50:07.719] mgos_pppos.c:867        PPPoS UART2 (RX:16 TX:17 CTS:14 RTS:15), 115200, fc on, APN 'everywhere'
[Feb 10 19:50:18.171] mgos e i 8    $  *'     i     _           S      Y}m{      mgos  .c:       $P: c ne  g
[Feb 10 19:51:30.973]  

Disabling PPPoS from the settings (by reflashing, mos config-set obviously stops working too) solves the issue.

rojer, 2019-02-10:
it looks like you are using the same uart for pppos and console? can you share your pppos settings?