mongoose-os-libs/pppos #6: SMS / Call

thbl, 2019-02-09:
Is it possible to send / receive sms / calls somehow while using pppos ?

rojer, 2019-02-09:
there is no API to send an SMS as such but there is an API to send custom commands which should work while ppp is active [ mgos_pppos_send_cmd()](https://github.com/mongoose-os-libs/pppos/blob/master/include/mgos_pppos.h#L51).

closed 2019-02-09 by rojer

thbl, 2019-02-09:
So with mgos_pppos_send_cmd we can execute normal AT commands ?
-I see in source it only reply true/false, is there a way to listen for the AT response ? (to check signal / check for new sms)

rojer, 2019-02-09:
the return value is actually just acnowledged that command was ququed, actual result is delivered via the MGOS_PPPOS_CMD_RESP event.