mongoose-os-libs/pppos #10: Unable to reset modem with rst settings

suyashmathema, 2019-06-18:
When I set these configs
[“pppos.rst_gpio”, 18]
[“pppos.rst_act”, 1]
[“pppos.rst_mode”, 1]
[“pppos.rst_hold_ms”, 1000]
[“pppos.rst_wait_ms”, 10000]

It never sets reset pin before pppos connection.

What I have found is that in the PPPOS_INIT state, initially net_status is set to MGOS_NET_EV_DISCONNECTED but before changing state to PPPOS_START_SEQ
this line is called `mgos_pppos_set_net_status(pd, MGOS_NET_EV_CONNECTING);`

The check in PPPOS_START_SEQ state always fails
`if (pd->net_status == MGOS_NET_EV_DISCONNECTED && (pd->cfg->rst_gpio >= 0 && (pd->attempt == 1 || pd->cfg->rst_mode == 1))) {
mgos_pppos_set_state(pd, PPPOS_RESET);
} else {
mgos_pppos_set_state(pd, PPPOS_BEGIN_WAIT);
}`

and it never reaches PPPOS_RESET state.

rojer, 2019-06-18:
i think you're right and it was a regression caused by https://github.com/mongoose-os-libs/pppos/commit/50ef31bc08927a19b80c97ca59f2b4812721430b
i'll send a fix

rojer, 2019-06-18:
try now

closed 2019-06-18 by rojer

suyashmathema, 2019-06-20:
yes it is working