mongoose-os-libs/pppos #14: PPPOS doesn't work with Esp32 TTGO T-Call

GiacomoGenovese, 2019-09-19:
Hi all,
I would use the TTGO T-call [https://github.com/Xinyuan-LilyGO/TTGO-T-Call] to create my GPS-GSM tracker. The board as a SIM800L chip integrated and I tried to use some code like the one prived by Alvaro @ https://github.com/alvarowolfx/asset-tracker-gcp-mongoose-os .
Anyway, I'm not able to connect to the mobile network and I keep getting this error:

"Serial console

20190730-142408/2.14.0-19-gb14a864-master

Ports available:
COM3
COM4

[Sep 19 19:45:22.018] mgos_pppos_uart_disp Command timed out: ATH
[Sep 19 19:45:22.024] mgos_pppos_uart_disp Connecting (UART1, APN 'wap.tim.it')...
[Sep 19 19:45:22.034] mgos_net_on_change_c PPP: disconnected
[Sep 19 19:45:24.448] mgos_net_on_change_c PPP: connecting
[Sep 19 19:45:27.527] mgos_pppos_uart_disp Command timed out: ATH
[Sep 19 19:45:27.532] mgos_pppos_uart_disp Connecting (UART1, APN 'wap.tim.it')...
[Sep 19 19:45:27.538] mgos_net_on_change_c PPP: disconnected
[Sep 19 19:45:28.220] mongoose_poll        New heap free LWM: 180240
[Sep 19 19:45:29.930] mgos_net_on_change_c PPP: connecting
[Sep 19 19:45:32.997] mgos_pppos_uart_disp Command timed out: ATH
[Sep 19 19:45:33.004] mgos_pppos_uart_disp Connecting (UART1, APN 'wap.tim.it')...
[Sep 19 19:45:33.007] mgos_net_on_change_c PPP: disconnected
[Sep 19 19:45:35.226] mgos_net_on_change_c PPP: connecting
[Sep 19 19:45:38.318] mgos_pppos_uart_disp Command timed out: ATH
[Sep 19 19:45:38.323] mgos_pppos_uart_disp Connecting (UART1, APN 'wap.tim.it')...
[Sep 19 19:45:38.328] mgos_net_on_change_c PPP: disconnected
[Sep 19 19:45:40.799] mgos_net_on_change_c PPP: connecting
[Sep 19 19:45:43.828] mgos_pppos_uart_disp Command timed out: ATH
[Sep 19 19:45:43.835] mgos_pppos_uart_disp Connecting (UART1, APN 'wap.tim.it')...
[Sep 19 19:45:43.839] mgos_net_on_change_c PPP: disconnected
[Sep 19 19:45:46.227] mgos_net_on_change_c PPP: connecting
[Sep 19 19:45:49.328] mgos_pppos_uart_disp Command timed out: ATH
[Sep 19 19:45:49.334] mgos_pppos_uart_disp Connecting (UART1, APN 'wap.tim.it')...
[Sep 19 19:45:49.337] mgos_net_on_change_c PPP: disconnected
[Sep 19 19:45:51.696] mgos_net_on_change_c PPP: connecting
[Sep 19 19:45:54.748] mgos_pppos_uart_disp Command timed out: ATH
[Sep 19 19:45:54.754] mgos_pppos_uart_disp Connecting (UART1, APN 'wap.tim.it')...
[Sep 19 19:45:54.758] mgos_net_on_change_c PPP: disconnected
[Sep 19 19:45:55.028] mongoose_poll        New heap free LWM: 180044
[Sep 19 19:45:57.027] mongoose_poll        New heap free LWM: 179828
[Sep 19 19:45:57.031] mgos_net_on_change_c PPP: connecting
........"

I also had a look to a similar issue in forum https://github.com/mongoose-os-libs/pppos/issues/2

Can you give me any help?

Thanks.

Bests.

rojer, 2019-09-20:
from the log i see that the chip cannot communicate with the modem. have you configured the UART pins [as specificed](https://github.com/Xinyuan-LilyGO/TTGO-T-Call#sim800-connection)? how do you configure modem?

GiacomoGenovese, 2019-09-26:
Thanks for your reply. We solved it with better pin configuration and a different antenna.

closed 2019-09-26 by GiacomoGenovese