mongoose-os-libs/pppos #25: u-blox nina lisa u200 and sara g350

tcpipchip, 2021-11-20:
Lisa u200 works perfectly on mongoose pppos :)
To make the Sara works i had to comment these line! Now working perfectly too :)
[image]

rojer, 2021-11-20:
interesting. AT+COPS operator selection is a standard command, i don't see why it wouldn't work.
can you run at debug.level=3 and post output?

tcpipchip, 2021-11-20:
hi, where i enable debug.level=3 ?
And do you want the debug with uncommented option, right ?

rojer, 2021-11-20:
> hi, where i enable debug.level=3 ?

add `- ["debug.level", 3]` to your app's conf_schema or after flashing, and assuming you have rpc-uart library,
run `mos config-set debug.level=3`

> And do you want the debug with uncommented option, right ?

yes, naturally.

tcpipchip, 2021-11-20:
attached
very weird
Attached
[capture.txt](https://github.com/mongoose-os-libs/pppos/files/7575031/capture.txt)
[capture2.txt](https://github.com/mongoose-os-libs/pppos/files/7575032/capture2.txt)
[capture3.txt](https://github.com/mongoose-os-libs/pppos/files/7575033/capture3.txt)

tcpipchip, 2021-11-22:
i see in the source code some comments about g350, then looks that already was tested on G350.

rojer, 2021-11-22:
that was before the quick operator reconnect was introduced, i didn't test it afterwards.

```
mgos_pppos.c:127        >> AT+COPS=0
mgos_pppos.c:137        8 -> 9 13.198152 613.194741
mgos_pppos.c:777        Command failed: AT+COPS=0
```

we don't log the response, really? dammit. can you hack the code a bit to print the response when at LL_DEBUG?

tcpipchip, 2021-11-22:
Hi @rojer
Weird that works perfectly with SARA G450 and LISA U200.
On G350 the AT+COPS=0 returns nothing...i tested on REALTERM...very weird! Maybe is G350 problem...
But if you send again AT+COPS=0 i finally gets a OK
Looks that there is not a TIME OUT implemented to AT+COPS=0 command...and stay in a LOOP.

tcpipchip, 2021-11-22:
btw, i love this PPPos, one of more stable that i have seen!

rojer, 2021-11-22:
indeed, weird. so maybe just implementing a retry of COPS command will do? kinda hacky, but not too difficult.

tcpipchip, 2021-11-22:
yes, should be good! Yes, i had tried to do ...but i am still not familiar with the CODE :)

tcpipchip, 2021-11-24:
@roger
let me know when have updates :)

rojer, 2021-11-24:
i checked my pile of stuff and i actually happen to have the 350 in there. when i get some spare time i'll set it up and see what's up.

tcpipchip, 2021-11-24:
Excelent!!! Excelent!!!!!!!!!!

tcpipchip, 2021-11-25:
If works at your side @rojer , then It´s a hardware problem in my side!